JeffersonLab/clas12-offline-software #1: Duplicate geometry classes

naharrison, 2017-04-25:
@drewkenjo I just noticed that some geometry classes exist twice in this repository. For example /common-tools/clas-geometry/src/main/java/org/jlab/detector/geant4/DCGeant4Factory.java and /common-tools/clas-jcsg/src/main/java/org/jlab/detector/geant4/v2/DCGeant4Factory.java. Could you remove any geometry stuff that is deprecated?

baltzell, 2018-03-20:
What's the status on this?  Open for almost a year now.

naharrison, 2018-03-21:
The duplicate classes still exist. Looks like all the reconstruction code is using org.jlab.detector.geant4.v2 (from clas-jcsg) so it should be safe to remove everything in org.jlab.detector.geant4 (from clas-geometry). Maybe @drewkenjo can confirm?

drewkenjo, 2018-03-21:
it requires removal of two packages: /common-tools/clas-geometry/src/main/java/org/jlab/detector/geant4/ and /common-tools/clas-geometry/src/main/java/org/jlab/geom/detector/ftof.
I was able to compile coatjava after I removed these two directories, so it seems that no other packages are using them. However, I was reluctant to do so because it could break other "external" packages that use COATJAVA. That's the usual problem with deprecated code, there is always some package that uses it. May be now is the best time to remove them, but we should be prepared that some packages might break. Let me know what you think.

raffaelladevita, 2018-03-21:
I would propose to send an email to clas12_software to announce these two packages will be removed in, let's say, two weeks and then proceed unless someone comes up with strong objections

forcar, 2018-03-21:
I use the detector.ftof package in some old DataProvider class that Gagik wrote, but that is probably expendable.

However parts of org.jlab.geom.detector.ec are used for in ECMON and the ECEngine and I would prefer that to remain for awhile.

Cole

heddle, 2018-03-21:
ced uses  /common-tools/clas-geometry/src/main/java/org/jlab/geom/
detector/ftof.

gavalian, 2018-03-21:
The geom.ftof and geom.ec are used in FAST MC, It wouldn’t be wise to remove them.

baltzell, 2018-03-21:
Sounds like this issue can be closed, and labeled "wontfix":)

heddle, 2018-03-22:
I think we should solve this problem of two geometry libraries. It goes
against everything we've tried to do with common tools. But we can't just
shut one down. We need to make sure that we create a union of the
functionality.

raffaelladevita, 2018-03-22:
It would be good to review what the differences are and what would be 
needed to converge on a single package.

drewkenjo, 2018-03-22:
Both package have a lot of overlapping functionalities actually. However, clas-jcsg was motivated when we needed an option to read STL files and import them to COATJAVA and GEMC. That was the main reason for clas-jcsg creation, and that's what it does. Now, for reconstruction it had to provide the functionalities of clas-geometry package but I couldn't simply reuse clas-geometry classes, because internally they are very different: the geometry objects in clas-jcsg are containers of CSG volumes, and all geometrical transformations and operations are done using CSG models. They also support hierarchical structure (mother/daughter volumes). So I had to create objects with similar functionalities.
If we want to unify them, some part of clas-geometry should be deprecated. The problem is that there are several large projects that rely on "deprecated" classes.
The best case scenario would be to parse these projects and come up with the list of methods they use, and them modify clas-jcsg to provide these methods. However, I don't have time do it alone, do we have manpower to dedicate for this project?

closed 2022-02-18 by baltzell