JeffersonLab/clas12-offline-software #91: change name and type of MC::Event

zhaozhiwen, 2018-03-23:
name and type of MC::Event at etc/bankdefs/hipo/MC.json are not general., thus force a generator to choose a limited format
How about change them to the following

            {"name":"npart",     "id":1,  "type":"int16", "info":"number of particles in the event"},
            {"name":"var1",   "id":2,  "type":"double", "info":"defined by generator"},
            {"name":"var2",   "id":3,  "type":"double", "info":"defined by generator"},
            {"name":"var3",   "id":4,  "type":"double", "info":"defined by generator"},
            {"name":"pbeam",     "id":5,  "type":"double", "info":"Beam polarization"},
            {"name":"var4",     "id":6,  "type":"double", "info":"defined by generator"},
            {"name":"var5",     "id":7,  "type":"double", "info":"defined by generator"},
            {"name":"var6",  "id":8,  "type":"double", "info":"defined by generator"},
            {"name":"var7", "id":9,  "type":"double", "info":"defined by generator"},
            {"name":"var8", "id":10, "type":"double", "info":"defined by generator"}

drewkenjo, 2018-03-24:
I'm not sure I understand what you propose. Do you want to remove all descriptive names, except npart and pbeam, from MC::Event, give them generic names "var1, var2, var3..." and make them double?

zhaozhiwen, 2018-03-24:
yes, this is one way to solve it.

raffaelladevita, 2018-03-24:
The variable naming of MC::Event was defined based on extensive discussions with many CLAS collaborators who defined a list of variables based on a "typical" event generator. Those btw matches the variables in the gemc Lund::header bank (see https://gemc.jlab.org/gemc/html/documentation/generator/lund.html). It is intended that user can use the variable to store different information if needed. In conclusion, changing the names to more generic ones would need a new discussion.

baltzell, 2019-05-02:
One year later and not enough motivation to change it, so shall we just close this?

closed 2019-05-04 by baltzell