JeffersonLab/clas12-offline-software #401: Throttle memory usage on decoder/evio2hipo

baltzell, 2019-09-23:
Currently decoder uses 3 GB on the JLab batch farm.  Is that really necessary, or can the current "-Xmx1536m -Xms1024m" be stricter?

gavalian, 2019-09-23:
After some tests I discovered that you can run decoder with -Xmx512m -Xms256m setting.
In this case the memory usage hovers around 700-710 MB, so setting 1GB limit should be fine.
However, lowering memory to -Xmx256 causes garbage collector to work harder and the decoding
slows down to half speed. So we can change decoder script to have maximum memory set to 512MB.

baltzell, 2019-09-23:
Might as well do the same for evio2hipo?

baltzell, 2019-10-29:
Done for evio2hipo, however the decoder effectively has some type of memory leak.

closed 2019-10-29 by baltzell