JeffersonLab/clas12-offline-software #427: Improve exception in DCHB

baltzell, 2019-12-01:
Avoid it, crash, or at least printout something to give the user some idea on what they're doing "wrong":

```
java.lang.NumberFormatException: For input string: "�"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Integer.parseInt(Integer.java:580)
	at java.lang.Integer.parseInt(Integer.java:615)
	at org.jlab.rec.dc.timetodistance.TableLoader.Fill(TableLoader.java:173)
	at org.jlab.service.dc.DCHBEngine.processDataEvent(DCHBEngine.java:108)
	at org.jlab.clas.reco.EngineProcessor.processEvent(EngineProcessor.java:242)
	at org.jlab.clas.reco.EngineProcessor.processFile(EngineProcessor.java:273)
	at org.jlab.clas.reco.EngineProcessor.main(EngineProcessor.java:368)

```

baltzell, 2020-03-18:
One way to reproduce it is just run recon-util on real data without a yaml.

raffaelladevita, 2022-05-19:
The exception is caused by incompatibility between the old and new time-to-distance . It will be resolved by making the polynomial function default

closed 2022-06-06 by zieglerv