JeffersonLab/clas12-offline-software #572: CCDB ConstantsManager improvements

baltzell, 2020-09-01:
* If a non-existing CCDB variation is specified, retries occur on every event, i.e. the mechanism that reports to Clara to abort on failed database connections isn't sensitive to non-existing variation
* Non-Clara usage does not abort in any case but should (decoder, evio2hipo, recon-util)

raffaelladevita, 2021-03-09:
- if non existent field map is specified, magfield gives exception at every event but Clara does not abort

baltzell, 2021-03-09:
Note the error catching in ReconstructionEngine:  https://github.com/JeffersonLab/clas12-offline-software/blob/development/common-tools/clas-reco/src/main/java/org/jlab/clas/reco/ReconstructionEngine.java#L256-L364

PR #677 (2021-03-11): Iss572 abort
Resolves #572
closed 2021-03-11 by raffaelladevita