JeffersonLab/clas12-offline-software #591: local cache for conditions database

sebouh137, 2020-10-01:
There should be a flag in recon-utils that allows the program to use a local cache of the conditions database, rather than connecting to the server every time.  This would be very useful when rapid-testing, debugging and development changes to the recon and other software, especially when running off-site.

baltzell, 2020-10-01:
See https://clasweb.jlab.org/wiki/index.php/CLAS12_Software_Center#tab=FAQ about getting an sqlite copy and using it by setting CCDB_CONNECTION/RCDB_CONNECTION variables.  Or see the environment modules for using a tagged one at JLab computers.  Or see the HOWTOs (on the same wiki page) on mounting CVMFS on your computer and reading a snapshot from there.

baltzell, 2020-10-01:
Also, note the forum is a good place to ask questions or search for existing answers.

closed 2020-10-01 by baltzell