JeffersonLab/clas12-offline-software #688: Magnetic Files not downloaded

gavalian, 2021-04-09:
It will be nice to have big warning message, preferably at the ent of build script (at the beginning it goes to fast and it's easy to miss) when magnetic filed maps are not successfully downloaded by build-coatjava script.
This can happen for many reasons (such as wget is missing from the system in my case), and user does not notice, and the crashes occur when trying to run recon-utils.

baltzell, 2021-04-09:
Can you check if branch iss688 addresses your issue?

gavalian, 2021-04-10:
It prints:
Retrieving field maps ...
./build-coatjava.sh: line 52: wget: command not found
ERROR:  Failed to download map:  Symm_solenoid_r601_phi1_z1201_13June2018.dat

However, what will happen if there is no connection to internet ? It will go through, but in the end it will not have the file.
I think it will make sense to have "test -e etc/data/magfiled/filename.dat || echo Attention The magnetic file is not present"

baltzell, 2021-04-10:
Do you know how to do a commit and pull request?

gavalian, 2021-04-10:
Come to think about it, may be the magnetic service can print a message that the file is not found. Instead of
stack trace ? The null pointer error is not descriptive enough.

closed 2021-04-10 by baltzell

gavalian, 2021-04-10:
The quoting is good for starters.

baltzell, 2021-04-10:
Cool