JeffersonLab/clas12-offline-software #738: More freehep maven repo problems

baltzell, 2021-11-04:
Probably offline?  This repo has been problematic in the past.  Previously we had a problem with this same dependency and http versus https and later maven versions, "fixed" via forcing it to https in [maven-settings.xml](https://github.com/jeffersonlab/clas12-offline-software/blob/development/maven-settings.xml).  

It gets pulled in through GROOT (which also currently won't build).

It's only instance in a pom in this repo is [common-tools/coat-lib/pom.xml](https://github.com/jeffersonlab/clas12-offline-software/blob/development/common-tools/coat-lib/pom.xml).

Only instance of freehep usage in this repo is in [RICH](https://github.com/jeffersonlab/clas12-offline-software/blob/development/reconstruction/rich), and it must be picking it up through [clas12maven](https://clasweb.jlab.org/clas12maven/org/freehep/jminuit/1.0/)?

baltzell, 2021-11-04:
Should we run our own maven repo for freehep on clasweb?

gavalian, 2021-11-04:
I tried to understand what is this about, and couldn't.
Why would we have to run repo for free hep ? What is the problem ?

baltzell, 2021-11-04:
Of course we don't *have* to, but it keeps being problematic.  Currently can't build coatjava or groot because of it.

gavalian, 2021-11-04:
Why can't we build GROOT ? I'm building it just fine.

baltzell, 2021-11-04:
Move your local m2 somewhere else and try.

baltzell, 2021-11-04:
hps-java still builds, looks like they've a different way of linking to freehep.

baltzell, 2021-11-04:
HPS does this instead, presumably that is more proper for this dependency (both are from SLAC):  https://github.com/JeffersonLab/hps-java/blob/47712878302eb0c0374d077a208a6f8f0e2c3dc6/pom.xml#L56-L60

That works for GROOT, but looks like we'd probably need a new GROOT release to pick it up for COATJAVA.

baltzell, 2021-11-04:
#740