JeffersonLab/clas12-offline-software #776: Add more (tagged) info banks to facilitate analysis

baltzell, 2022-01-14:
For discussion.  Some suggestions from Derek below.  One caveat is if people merge data across multiple conditions.
* CCDB stuff
  * beam energy (still not in CCDB?)
  * target position
  * RF bucket length
  * ECAL sampling fraction parameters
* run period (e.g. "fall_2018")
* cooking pass (e.g. "pass1")
* simulation vs real data flag
  * RUN::config.type/mode is already available for that
    * but not ever tagged currently in MC (and maybe not even set?)
  * there's also supposed to soon be tagged configuration banks from GEMC that would also serve as a flag
    * and would also contain some of the parameters above

dglazier, 2022-01-14:
test

dglazier, 2022-01-14:
test

baltzell, 2022-12-07:
Including CCDB tables would require new version of HIPO data format.  Software versionings are already included.  Closing this for now.

closed 2022-12-07 by baltzell