JeffersonLab/clas12-offline-software #805: DC track fit residual parallax

zieglerv, 2022-03-09:
Resolve DC track fit residual parallax resulting from plane rotations and tilts not accounted in tracking

baltzell, 2022-12-07:
Is this still valid?

tongtongcao, 2022-12-07:
I think that geometry alignment is not taken into account in current forward tracking.
When we develop track representation with 6 parameters as reference of path length in the future, plane rotations and tilts will be taken care of, and this issue will be solved by then.

raffaelladevita, 2022-12-07:
6-dimensional track representation is indeed necessary to support full alignment, with all translations and rotations. However, the current representation does support all translations and the z-rotation in the tilted frame. Tests with MC simulations, where known misalignments were introduced, showed that y rotation is also partially accounted for, while the y rotations cannot be corrected for. See 15 May 5, 2022 <https://clasweb.jlab.org/wiki/index.php/Clas12_software_meetings_2022#May_5.2C_2022> and following meetings.