JeffersonLab/clas12-offline-software #840: EC Last-a-Foam Thickness Should be 76 mm

forcar, 2022-05-16:
Original specifications for EC had Last-a-Foam thickness=76 mm, which is consistent with design drawings and external measurements of the EC lid thickness.   

Releases of GEMC prior to 2.4  used 76.2 mm, which is still in geometry.pl:  https://github.com/gemc/detectors/blob/master/clas12/ec/geometry.pl

Current tagged versions show 25.4 mm for the G4Trap half-width: https://github.com/gemc/clas12Tags/blob/master/4.3.0/experiments/clas12/ec/ec__geometry_java.txt

baltzell, 2022-12-07:
What will it take to make this happen?

baltzell, 2023-01-19:
Do we need this for pass2?

raffaelladevita, 2023-01-19:
Cole should confirm but I believe it affects only the simulation geometry. So it is not relevant for pass2 but should still be fixed

forcar, 2023-01-19:
It is intended to make simulations of e- and gamma sampling fractions more realistic. Recall photon SF correction is derived from MC.

closed 2023-03-28 by raffaelladevita