JeffersonLab/clas12-offline-software #899: DataSourceProcessorPane doesn't allow setting the verbosity of the underlying DataSourceProcessor

raffaelladevita, 2022-07-04:
All applications using the DataSourceProcessorPane, like mon12 or the calibration suites, do not report exceptions in the event processing because these are caught at https://github.com/JeffersonLab/clas12-offline-software/blob/fe68c2bc773aec704837ee6c7f7d301eae2ebd0f/common-tools/clas-io/src/main/java/org/jlab/io/task/DataSourceProcessor.java#L137 
This makes identifying problems difficult. 
A method to set the verbosity of the DataSourceProcessor should be added to the DataSourceProcessorPane

baltzell, 2022-07-04:
I guess maybe there was a good reason not to just print all exceptions?

baltzell, 2022-12-07:
Is this still relevant?

raffaelladevita, 2022-12-07:
Yes, it is. It’s a small change that should be done to facilitate debugging calibration code

baltzell, 2022-12-07:
Maybe you just want this in the catch:
System.out.println(e.getMessage());

baltzell, 2023-03-16:
closed by #1087

closed 2023-03-16 by baltzell