JeffersonLab/clas12-offline-software #984: ECAL use of tdcs map is not thread safe

raffaelladevita, 2022-11-20:
use of static tdcs map at https://github.com/JeffersonLab/clas12-offline-software/blob/development/reconstruction/ec/src/main/java/org/jlab/service/ec/ECCommon.java#L334-L337 should be disabled because gives multi-threading conflicts. Since it seems it is not used in reconstruction anyway, its use could be limited as it is done for the H1F collections

raffaelladevita, 2022-11-20:
Also get this exception when running on data with Clara:
Error processing input event

java.lang.IndexOutOfBoundsException: Index -1 out of bounds for length 6
        at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
        at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
        at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
        at java.base/java.util.Objects.checkIndex(Objects.java:372)
        at java.base/java.util.ArrayList.get(ArrayList.java:458)
        at org.jlab.service.ec.ECPeak.splitPeak(ECPeak.java:483)
        at org.jlab.service.ec.ECPeakAnalysis.splitPeaks(ECPeakAnalysis.java:36)
        at org.jlab.service.ec.ECCommon.processPeaks(ECCommon.java:412)
        at org.jlab.service.ec.ECEngine.processDataEvent(ECEngine.java:36)
        at org.jlab.clas.reco.ReconstructionEngine.execute(ReconstructionEngine.java:361)
        at org.jlab.clara.sys.ServiceEngine.executeEngine(ServiceEngine.java:227)
        at org.jlab.clara.sys.ServiceEngine.execute(ServiceEngine.java:153)
        at org.jlab.clara.sys.Service.lambda$execute$2(Service.java:178)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)

forcar, 2022-11-20:
The tdcs map has been in ECCommon for all pass1 cooking.  Why is it causing problems now?

baltzell, 2022-11-20:
Possibly some things went under the radar in pass1.  It's either thread-safe or it's not.  If we now know it's not, it should be fixed or just not used in pass2.

forcar, 2022-11-20:
I see what happened.  I moved the instantiation from the beginning of readStripsHipo to the header and made it static.   IndexedList is a gagik class but it worked fine as it was.  Let's revert back. 

[image]

[image]

forcar, 2022-11-21:
Branch iss984-ecal-static-issues has removed the offending static references but won't build due to org.freehep Bad Gateway

[image]

raffaelladevita, 2022-11-30:
addressed by #986

closed 2022-11-30 by raffaelladevita