JeffersonLab/clas12-offline-software #1017: New exception in DC engine

baltzell, 2022-12-18:
It's happening in multiple chefs' logs:

```
java.lang.ArrayIndexOutOfBoundsException: Index 3 out of bounds for length 3
    at org.jlab.service.dc.DCTBEngine.processDataEvent(DCTBEngine.java:202)
    at org.jlab.clas.reco.ReconstructionEngine.execute(ReconstructionEngine.java:376)
    at org.jlab.clara.sys.ServiceEngine.executeEngine(ServiceEngine.java:227)
    at org.jlab.clara.sys.ServiceEngine.execute(ServiceEngine.java:153)
    at org.jlab.clara.sys.Service.lambda$execute$2(Service.java:178)
    at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
    at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
    at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
    at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
    at java.base/java.lang.Thread.run(Thread.java:834)
```

So, at line 202 here:

https://github.com/JeffersonLab/clas12-offline-software/blob/824f31685794aedf6e1a8b433c8cc7f2e29cecfc/reconstruction/dc/src/main/java/org/jlab/service/dc/DCTBEngine.java#L200-L206

baltzell, 2023-01-19:
Has this been addressed?

raffaelladevita, 2023-02-24:
Addressed by #1041

closed 2023-02-24 by raffaelladevita