JeffersonLab/clas12-offline-software #1034: RICH is throwing exceptions

baltzell, 2023-01-27:
Looks like there's an issue in RICH reconstruction in 8.5.0, at least for RG-C run number 16189.  It spews this exception on pretty much every event.  YAML file that was used is attached.

```
java.lang.NullPointerException
	at org.jlab.detector.geom.RICH.RICHGeoFactory.get_Pixel_Center(RICHGeoFactory.java:596)
	at org.jlab.rec.rich.RICHHit.<init>(RICHHit.java:85)
	at org.jlab.rec.rich.RICHPMTReconstruction.reco_PMTHits(RICHPMTReconstruction.java:232)
	at org.jlab.rec.rich.RICHPMTReconstruction.process_RawData(RICHPMTReconstruction.java:63)
	at org.jlab.rec.rich.RICHEBEngine.processDataEvent(RICHEBEngine.java:139)
	at org.jlab.clas.reco.ReconstructionEngine.execute(ReconstructionEngine.java:376)
	at org.jlab.clara.sys.ServiceEngine.executeEngine(ServiceEngine.java:227)
	at org.jlab.clara.sys.ServiceEngine.execute(ServiceEngine.java:153)
	at org.jlab.clara.sys.Service.lambda$execute$2(Service.java:178)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
```

[yaml.txt](https://github.com/JeffersonLab/clas12-offline-software/files/10520116/yaml.txt)

baltzell, 2023-01-27:
FYI, at least RG-B doesn't appear to have this problem.

baltzell, 2023-02-24:
I think this was supposed to have been addressed by #1037?

raffaelladevita, 2023-02-24:
Yes, correct

closed 2023-02-24 by raffaelladevita