JeffersonLab/clas12-offline-software #1050: RICH bank name changes need updating in schema generator script

baltzell, 2023-02-21:


baltzell, 2023-02-21:
For example, the current RICH banks at:

https://github.com/JeffersonLab/clas12-offline-software/blob/development/etc/bankdefs/hipo4/rich.json

(RICH::Cluster,Hit,Signal,Response,Hadron,Photon,Ring,Particle)

don't jive with what's in the schema generator script:

https://github.com/JeffersonLab/clas12-offline-software/blob/f4ac4b730844294adde0bddb9cb2fe644122b36d/etc/bankdefs/util/bankSplit.py#L69

(RICH::hits,hadCher,hadrons,photons,ringCher)

Some look like an obvious 1:1 translation, some don't.

mcontalb, 2023-02-22:
The new banks in the calibration skim are
"RICH::tdc","RICH::Hit","RICH::Particle","RICH::Hadron","RICH::Photon","RICH::Ring"
Not sure if I have to do a new pull request for the fix

baltzell, 2023-02-22:
Someone has to define the mapping between the old bank names that remain in the generator script, and the new bank names.  Most are obvious, some aren't.

mcontalb, 2023-02-22:
Not sure what is the generator script, but the mapping is

RICH::tic -> RICH::tdc
RICH::hits -> RICH::Hit
RICH::hadCher -> RICH::Particle
RICH::hadrons -> RICH::Hadron
RICH::photons -> RICH::Photon
RICH::ringCher -> RICH::Ring

If I have to do the change, I would need guidance how to do it.

mcontalb, 2023-02-22:
Sorry: RICH::tdc -> RICH::tic

PR #1058 (2023-02-23): Fix RICH banks for RICH
closes #1050
closed 2023-02-23 by raffaelladevita